SpellFoundry/SleepyPi2 #3: WakePiPeriodicallyOnAlarm.ino Not Working As Expected

kiteboy, 2017-08-24:
Hi looking for advice really

Im trying to start the Pi at say 3pm and every 30mins wake up and do its thing and then shutdown
I have the sleepy pi 2 starting every 30mins fine but not on or close to the 30min mark ie 3pm ......3.30pm.....4pm
All I can seem to do is start every 30min and that depends on the compilatiion time itself

Can anybody help please this is what I get in the serial monitor and in this exammple it would wake up and start again att 15:23 say

`Starting, but I'm going to go to sleep for a while...
Ok, Time = 14:53:40, Date (D/M/Y) = 24/8/2017
Alarm Set for every: 30 minutes
`
The code im using is:

Can anybody see anything wrong------ so close

//Simple example showing how to set the RTC alarm pin to periodically wake up the Arduino
// which in turn wakes up the Rpi. You can use this to do things with the Rpi 
// at regular intervals i.e. like taking a photograph.
//
// Note, in this example the Rpi is woken up for every 5 minutes (you can change that).
// Once woken it is then shutdown after 1 minute (you can change that) unless the user 
// hasn't already shut the Rpi down. The system then goes into low power mode until 
// the next scheduled time interval to wakeup.
//
// This example shows an alternative method than the Timer to do periodic wakeups.
//
// To test on the RPi without power cycling and using the Arduino IDE
// to view the debug messages, comment out these line (with a //):
//
// SleepyPi.enablePiPower(true);
// SleepyPi.piShutdown();      
// SleepyPi.enableExtPower(false);  
// 
// Also either fit the Power Jumper or enable
// self-power. http://spellfoundry.com/sleepy-pi/programming-arduino-ide/
// 

// **** INCLUDES *****
#include <SleepyPi2.h>
#include <Time.h>
#include <TimeLib.h>
#include <LowPower.h>
#include <PCF8523.h>
#include <Wire.h>

const char *monthName[12] = {
  "Jan", "Feb", "Mar", "Apr", "May", "Jun",
  "Jul", "Aug", "Sep", "Oct", "Nov", "Dec"
}; 

const int LED_PIN = 13;

// Globals
// ++++++++++++++++++++ CHANGE ME ++++++++++++++++++
uint8_t  WakeUp_StartHour     = 15;   // Hour in 24 hour clock
uint8_t  WakeUp_StartMinute   = 30;   // Minutes 

unsigned long    MAX_RPI_TIME_TO_STAY_AWAKE_MS  = 60000;       // in ms - so this is 60 seconds
#define kPI_CURRENT_THRESHOLD_MA   110                         // Shutdown current threshold in mA. When the
                                                               // when the Rpi is below this, it is "shutdown"
                                                               // This will vary from Rpi model to Rpi model
                                                               // and you will need to fine tune it for each Rpi
                                                               // See Code below for tuning code
// ++++++++++++++++++++ END CHANGE ME ++++++++++++++++++

tmElements_t tm;
uint8_t   nextWakeTime;

void alarm_isr()
[... 240 lines cut ...]
      Serial.println(reg_value,HEX);      
      reg_value = SleepyPi.rtcReadReg(PCF8523_WEEKDAY_ALARM);
      Serial.print("Weekday Alarm: ");
      Serial.println(reg_value,HEX); 
      
      reg_value = SleepyPi.rtcReadReg(PCF8523_OFFSET);
      Serial.print("Offset: 0x");
      Serial.println(reg_value,HEX); 
      reg_value = SleepyPi.rtcReadReg(PCF8523_TMR_CLKOUT_CTRL);
      Serial.print("TMR_CLKOUT_CTRL: 0x");
      Serial.println(reg_value,HEX);  
      reg_value = SleepyPi.rtcReadReg(PCF8523_TMR_A_FREQ_CTRL);
      Serial.print("TMR_A_FREQ_CTRL: 0x");
      Serial.println(reg_value,HEX); 
      reg_value = SleepyPi.rtcReadReg(PCF8523_TMR_A_REG);
      Serial.print("TMR_A_REG: 0x");
      Serial.println(reg_value,HEX);     
      reg_value = SleepyPi.rtcReadReg(PCF8523_TMR_B_FREQ_CTRL);
      Serial.print("TMR_B_FREQ_CTRL: 0x");
      Serial.println(reg_value,HEX);    
       reg_value = SleepyPi.rtcReadReg(PCF8523_TMR_B_REG);
      Serial.print("TMR_B_REG: 0x");
      Serial.println(reg_value,HEX);     
 
}'

daghemo, 2018-09-21:
Hi @kiteboy,
mee too I'm interested in this! Have you found a solution?

kiteboy, 2018-09-26:
No sorry - didnt exactly get around that issue.  Very few support avenues too. If you find any process of doing it let me know

daghemo, 2018-09-27:
I understand. Still waiting for an anwer from @SpellFoundry on another theme. A good board, what a pity.