SpellFoundry/SleepyPi2 #6: SleepyPi board on Arduino IDE

Joselin15, 2018-03-20:
Hi everyone,

I am trying to connect my Sleepy Pi 2 board in the Arduino IDE but after I go the IDE > Tools > Board  it does not show up the name of the Sleepy Pi. Also when I tried to see the examples from the Sketchbook library, I cannot seem to find it at all. Is there any additional step I need to take to see my board.
I installed the Sleepy-Pi-Setup.sh and configure the clock so that both RTC and Raspberry Pi are synchronized correctly with the date and time.
I hope anyone can help me.

Thanks.

jensjakob, 2018-03-20:
I would check to see if the files is installed for the right users. You might have installed it for another user than default.

jensjakob, 2018-03-23:
The Sleepy-Pi-Setup.sh script places the files in the folder for the default user for Raspberry Pi. The default user is "pi" and the files is placed at: /home/pi/sketchbook

Hope this will help!

I think I had the same problem, so it would be better to be prompted for the user or placed it in the current users home directory. Right? But the installation is for the user "pi", only.

Bartal92, 2018-03-26:
Hi, Joselin
I don't know if Jens's answers fixed your problem, but I had a similar problem. I didn't use the script, because I'm running Raspbian (stretch), and the script doesn't work for that. And the way described in https://spellfoundry.com/sleepy-pi/setting-arduino-ide-raspbian/ didn't work. What I did to fix it was I copied the text, that the link says should be in the boards.txt, and found the original boards.txt, were the other arduino boards are also described, in the arduino folder, and opened it and at the end I pasted the text from https://spellfoundry.com/sleepy-pi/setting-arduino-ide-raspbian/ in that file. That worked for me.

peggyopal, 2018-05-25:
@Bartal92 Did you have to do anything extra? I did the same thing as you but I'm still getting the error:

`FILENAME.ino:8:23: fatal error: SleepyPi2.h: No such file or directory
compilation terminated.`

tennisparty, 2019-10-26:
Hello all,

I am having very similar problems. 

I am using a Pi zero W. I am trying to get the blink script to work.

Based on your advice, I have tried changing the Arduino IDE preferences to /home/pi/sketchbook and also moving the boards.txt file into /home/pi/arduino-1.8.10/hardware/arduino/avr
I can now select the Sleepy Pi as a board but I get the following error message when uploading (and have tried through both available ports - dev/ttyAMA0 and dev/ttyS0)

> "Warning: Board arduino:avr:sleepypi doesn't define a 'build.board' preference. Auto-set to: AVR_SLEEPYPI
> ......
> Error while uploading: missing 'upload.tool' configuration parameter"

I have tried powering the setup in several ways. Into both the sleepy pi and the pi W micro USB's or either or (I read elsewhere it could be a power issue)

Any more ideas?
Sam

tennisparty, 2019-10-28:
I got it working (lost a day of the weekend though!). I had to manually add 'enable_uart=1' to the bottom of the boot file (am unsure why this didn't work in the script). It then works if I use port dev/ttyS0 (I imagine because the pi zero W has a bluetooth chip). I then need to press the reset button a few times whilst the code upload starts. Success!