SpellFoundry/SleepyPi2 #10: Arduino on the Sleepy Pi 2 restarting over and over

daghemo, 2018-10-08:
The Arduino on my Sleepy Pi 2 is restarting over and over after some time with apparently no reason. 
I've also tried my other Sleepy Pi 2 and even a different Raspberry Pi.

The following code can be used to reproduce the error:

```
#include "SleepyPi2.h"

unsigned long timeNow = 0, timeLast = 0;
char datetime[ 31 ];

void setup() {
    Serial.begin( 9600 );
    Serial.println( "Starting serial console..." );
    SleepyPi.rtcInit( true );
}

void loop() {
    DateTime now;
    timeNow = millis();
    if ( ( timeNow - timeLast ) >= 500 ) {
        now = SleepyPi.readTime();
        sprintf( datetime, "%d/%02d/%02d %02d:%02d:%02d %lu" , now.year(), now.month(), now.day(), now.hour(), now.minute(), now.second(), timeNow );
        Serial.println( datetime );
        Serial.flush();
        timeLast = timeNow;
    }
}
```

The above code has been slightly modified from #9.
On the serial I see the following output (just a sample):

```
2018/10/08 13:33:15 32024
2018/10/08 13:33:16 32524
2018/10/08 13:33:16 33024
2018/10/08 13:33:17 33525
Starting serial console...
2018/10/08 13:33:18 501
2018/10/08 13:33:18 1001
2018/10/08 13:33:19 1501
2018/10/08 13:33:19 2002
Starting serial console...
2018/10/08 13:33:20 501
2018/10/08 13:33:20 1001
2018/10/08 13:33:21 1501
2018/10/08 13:33:21 2002

```
Have you got any ideas?

daghemo, 2018-10-08:
It seems that running "i2cdetect -a -y 1" will cause the Arduino to restart.
This seems also to be true for any code enumerating devices on the I2C bus.

SpellFoundry, 2018-10-15:
if you leave out the "-a" then the reset doesn't occur. According to the Linux manual:

"[-a] Force scanning of non-regular addresses. Not recommended."

daghemo, 2018-10-16:
Removing the "-a" seems to solve the problem!
Looking at the sources of i2cdetect I can see that not using "-a" means checking only addresses from 0x03 to 0x77 so I've modified my code to to that.

closed 2018-10-23 by SpellFoundry