SpellFoundry/SleepyPi2 #11: Arduino on the Sleepy Pi 2 unresponsive/halted

daghemo, 2018-10-23:
I've uploaded a Sketch to the Arduino on my Sleepy Pi 2.
It wakes up every 10 seconds to print something (date, current, voltage and other) to the serial, but it also wakes up by alarm twice a day, to punch in in the morning and punch out in the evening. The Raspberry is turned on on the first alarm and off on the second one. After every alarm I just powerDown(). 
The Sketch has been used on two boards for days.

Yesterday one of my boards did not went off in the evening. The time on the board was correct, aligned with the one from the RTC. But I was not able to see something on the serial. That is, the Arduino seemed freezed.

Any suggestions? Is there a way I can prevent this from happening or just a way to reset/refresh the Arduino?

Please, be gentle: I'm not an Arduino expert! ;)

SpellFoundry, 2018-10-23:
Difficult to tell but this may be the Arduino library i2c having some infinite while loops: https://github.com/arduino/Arduino/issues/1476
You might want to try some of the fixes around the web such as this one: https://rheingoldheavy.com/changing-the-i2c-library/

daghemo, 2018-10-24:
I do not use I2C directly in my Sketch.
Do you think it could be your own implementation of PCF8523::readTime()? I now it uses the Wire library.

Any proposal to refresh/reset the Arduino on the Sleepy Pi 2, with no impact on the Raspberry Pi? That is, without powering it down.